randdusing/ng-cordova-bluetoothle #2: Can't seem to get scan results on Nexus 5

agoransson, 2015-11-09:
I'm running this example just fine, on the surface everything seems to work just fine. Init works, enable/disable works, scan starts and stops, getting connected devices works just fine.

My problem is that scan results are not returning to the callback. I can see through logcat that the scan is successfully finding the device I'm trying to connect to (TI SensorTag), but the app does not record any found devices.

Do you know what could be the problem?

I've not modified the code in any way.

randdusing, 2015-11-10:
What version of Android are you on? I'm having the same issue with an app user of mine. He's on a Nexus 5 and Android 6.0. He can see his device within other apps and from Bluetooth settings.

Any way you could add a log statement to the top of this function just to see how often it's called?
```java
public void onLeScan(final BluetoothDevice device, int rssi, byte[] scanRecord)
```

Assuming it's getting called for each scan result you see in the Bluetooth log, the scanCallbackContext must be getting set to null, otherwise I think you'd see an exception.

agoransson, 2015-11-10:
The device is running android 6 (MRA58N).

I added a log out at the top of that method, but it doesn't seem to be called at all. I can still see the scan finding them in the log cat though.

randdusing, 2015-11-10:
@agoransson See this: https://github.com/randdusing/BluetoothLE/issues/204 Do you see anything in the log about not having the proper permissions?

agoransson, 2015-11-10:
I just saw that issue in your main repo - perhaps this discussion should be closed and I'll move on to that issue instead - I'll close this.

No permission messages related to bluetooth - just some random gms and music permissions whining.

closed 2015-11-10 by agoransson