randdusing/ng-cordova-bluetoothle #3: What should be the format of the uuid string in the service and characteristic in a write request?

NitishDeshpande, 2016-01-27:
Hi I am trying to send a message to a BLE node using GATT connection. I am able to establish a connection but I am not able to send a message. This is the error I am facing,
Write Error : 
{"error":"service","message":"Service not found","address":"C2:A7:51:2C:F2:37","name":"ble-Micro"}
My request looks like this,
```
$scope.write =function() {
    	    var params = {
                         address:"C2:A7:51:2C:F1:47", 
                         service:"713d0003-503e-4c75-ba94-3148f18d941e", 
                         characteristic: "713d0003-503e-4c75-ba94-3148f18d941e", 
                         value:"hello world", 
                         timeout: 3000
                       };
    	    console.log("Write : " + JSON.stringify(params));

    	    $cordovaBluetoothLE.write(params).then(function(obj) {
    	      console.log("Write Success : " + JSON.stringify(obj));
    	    }, function(obj) {
    	      console.log("Write Error : " + JSON.stringify(obj));
    	    });
 };```

Any kind of help is appreciated.

randdusing, 2016-01-27:
Can you post the rest of your code? And what version of the plugin are you using?

NitishDeshpande, 2016-01-28:
I am using version 3.0.1. 

Here is rest of the code which initializes and connects before writing/or reading,
```
    $scope.initialize = function(){
    	  $cordovaBluetoothLE.initialize({request:true}).then(null,
    			    function(obj) {
    			      //Handle errors
    		  		console.log(obj);
    			    },
    			    function(obj) {
    			      //Handle successes
    			    	console.log('success');
    			    }
    			  );
    };

      $scope.connect = function(){
    	  $cordovaBluetoothLE.connect({address:"C2:A7:51:2C:F1:47"}).then(null,
  			    function(obj) {
  			      //Handle errors
  			      console.log(obj.message);
  			    },
  			    function(obj) {
  			      if (obj.status == "connected") {
  			    	  console.log(obj);
  			        console.log('Connected');
  			      } else {
  			    	  console.log('Disconnected');
  			      }
  			    }
  		);  
      };

$scope.write =function() {
            var params = {
                         address:"C2:A7:51:2C:F1:47", 
                         service:"713d0000-503e-4c75-ba94-3148f18d941e", 
                         characteristic: "713d0003-503e-4c75-ba94-3148f18d941e", 
                         value:"hello world", 
                         timeout: 3000
                       };
            console.log("Write : " + JSON.stringify(params));

            $cordovaBluetoothLE.write(params).then(function(obj) {
              console.log("Write Success : " + JSON.stringify(obj));
            }, function(obj) {
              console.log("Write Error : " + JSON.stringify(obj));
            });
 };

```

NitishDeshpande, 2016-01-28:
I am getting a similar error when I am subscribing, (console logs shown below)
```
Request-
Subscribe : {"address":"C2:A7:51:2C:F1:47","service":"713d0000-503e-4c75-ba94-3148f18d941e","characteristic":"713d0002-503e-4c75-ba94-3148f18d941e","timeout":2000}

Subscribe Error : {"error":"service","message":"Service not found","address":"C2:A7:51:2C:F1:47","name":"ble-Micro"}
```

randdusing, 2016-01-28:
You need to discover the device first. Please check out the plugin's readme: https://github.com/randdusing/cordova-plugin-bluetoothle

NitishDeshpande, 2016-01-28:
Thanks for quick response. Discovery is done mainly to determine the address of the device, the services available and their corresponding characteristic. I already know the address, service uuid and characteristic uuid using another android app (BLE Scanner). Shouldn't this be fine?

randdusing, 2016-01-28:
Unfortunately no, you must discover the device even if you already know what's on it.

NitishDeshpande, 2016-01-28:
Ohh okay! Will try it and let you know :)

NitishDeshpande, 2016-01-28:
I tried the discovery method before calling the write/subscribe methods.
- When I try to write something, the app closes abruptly.
- Subscribe works fine. What is format of the data that is received?
When a character "a" is sent from the BLE device I am receiving "YQ0K" on my mobile app

randdusing, 2016-01-28:
Can you post the error message when the app closes? Check out the encodedStringToBytes and bytesToString functions to decode the base64 encoded string.

NitishDeshpande, 2016-02-08:
@randdusing Thanks for your support. I was able to do the subscribe() and write() operations successfully. I have used the methods to convert string to bytes and bytes to encoded string and other utility functions. All of these work as expected without any issues. Hence, closing this issue.

closed 2016-02-08 by NitishDeshpande