randdusing/ng-cordova-bluetoothle #4: Missing template files?

majorbytes, 2016-02-11:
Where are these located:
<script type=text/ng-template id=home.html>
<script type=text/ng-template id=device.html>
<script type=text/ng-template id=service.html>
<script type=text/ng-template id=characteristic.html>

randdusing, 2016-02-11:
The templates within the index.html file in the example.

closed 2016-02-11 by randdusing

majorbytes, 2016-02-11:
It comes up all wonky in android on the screen..

randdusing, 2016-02-11:
Wonky? Can you take a screenshot

majorbytes, 2016-02-11:
http://topofire.dbs.umt.edu/tmp/BluetoothScreenshot.png

randdusing, 2016-02-11:
It looks fine to me. What issue are you having?

majorbytes, 2016-02-11:
Ok, if that is what it is supposed to look like...
Sorry, I was expecting tabbed pages or something..

randdusing, 2016-02-11:
Yah, it's just an example :)

majorbytes, 2016-02-11:
Thanks...  Trying to write a frontend to a Kestrel weather device and i found your github..