randdusing/ng-cordova-bluetoothle #5: Cordova BluetoothLE Detects only mac laptop no other devices?

Kailash43, 2016-02-12:
I was trying to figure out how BLE works. Tried to go through your plugin have followed all the steps mentioned in document. Build for android using windows everything is working great till now.

- Initialize my device.
- Enable/Disable my device bluetooth.

But when i used to startScan it only detects my Mac laptop even if my all other devices are enabled. I wanna know 

- How the app scan for all the bluetooth devices?
- What should i have to pass in params  of startScan show that it detects all the devices?

randdusing, 2016-02-12:
Are the devices advertising in Bluetooth LE mode? The scan only picks up Bluetooth LE devices.

Kailash43, 2016-02-13:
Build bluetoothLE app on One plus 2 and trying to scan mac, samsung galaxy s6 and one plus x device but there is no result execpt mac laptop.

randdusing, 2016-02-13:
Are they advertising as Bluetooth LE devices? They most likely aren't, which is why the scan doesn't seem them.

Kailash43, 2016-02-14:
Samsung Galaxy S6 and Oneplus X specs shows that it supports Bluetooth 4.0 (LE), however the bluetooth plugin isn’t able to discover those devices. Is there any further configuration that I need to do to enable its discovery? Please let me know.

randdusing, 2016-02-14:
Your devices need to be advertising as Bluetooth LE devices. You can use an app like LightBlue to accomplish this.

closed 2016-02-14 by randdusing