randdusing/ng-cordova-bluetoothle #6: Read function not workind

wifixcort, 2016-02-15:
Hi I'm using an Bluetooth  from Adrafuit(Bluefruit). I can write without problem from my Android phone but I can't read anything form bluefruit. I just got errors(Unable to read).

Adafruit bluefruit https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/uart-service

I use the example code adapted to a simple application without modification.

I have noticed something very simple, it takes a moment to write and displays "Success" or "Errors". But to read almost immediately displays "Error"

This is my test project https://github.com/wifixcort/BlueTest

Thanks for help

randdusing, 2016-02-16:
Do you have any other Bluetooth LE devices you could test the read to? If the app code is nearly the same as my example, it may be an issue with the Adrafruit.

wifixcort, 2016-02-16:
:'( sadly no, this is the one I have to use in the application that I need to develop

wifixcort, 2016-02-16:
In your example code I can't see a notification function, there is a way to make one?

randdusing, 2016-02-17:
Without the device to test reading with, I don't think I'll be able to help much. For notification, check out the subscribe function.

ricardoMogg, 2016-02-19:
One way to test this easily might be using an ios (i'm working with an ipad mini 8.1.2)with the app light blue... this has been working for me.

randdusing, 2016-02-20:
Yah the LightBlue app is very helpful for emulating other types for BLE devices

randdusing, 2016-03-23:
You can also use v4.4.0 of the plugin and wrapper to easily create a peripheral on your Android or iOS device. Please reopen if you continue having the issue.

closed 2016-03-23 by randdusing