randdusing/ng-cordova-bluetoothle #8: Read works only after discover function ended

papasmurf17, 2016-03-01:
Is there any reason why the function **read** works only if executed inside the callback of function **discover** ?

randdusing, 2016-03-01:
Yah, you can't read/write/subscribe until discovery has completed. Read doesn't necessarily need to be inside the discover callback, but it needs to be called after discovery has completed.... minor distinction.

papasmurf17, 2016-03-01:
Thanks for the explanation!

I've another question, I'm developing my first app with bluetooth features so I'm a noob.
If I need send a command to bluetooth device and after read the response, I need subscribe a service, write the command on correct service/characteristic and wait the response?

I'm sorry if it's a stupid question

randdusing, 2016-03-01:
No worries, that's good approach. Subscribe, write command, and then wait for response.

papasmurf17, 2016-03-01:
Thank you so much 😊

closed 2016-03-23 by randdusing