randdusing/ng-cordova-bluetoothle #9: Can't get example app to work

struempelix, 2016-03-14:
Hi Rand, 
for some reason I can't get your example app to work. It shows the "Central role" screen only, but I'm sure I've previously seen the footer with "Peripherals" & "Log" as well. Also, non of the buttons triggers an action. I'm working on Android. Any guess what I've might done wrong?

Thanks,
Jens

[image]

struempelix, 2016-03-14:
Arg, sorry, posted to early. I had mixed up default and dev branch. Now it is working! Thanks for your work on this!

randdusing, 2016-03-14:
No problem, glad you got it working!

closed 2016-03-14 by randdusing