randdusing/ng-cordova-bluetoothle #14: Long time to disconnect/close ble connection for iOS

MuriloACQ, 2016-04-30:
When I send disconnect/close command in Android platform the delay is very short, but in iOS it takes about 10 seconds to disconnect. Is there a way to disconnect faster in iOS?

randdusing, 2016-04-30:
That's strange. Disconnects have been pretty quick for me on both Android and iOS. What type of device are you connecting to?

MuriloACQ, 2016-04-30:
I was connecting  iPhone 5c to HM-10 module. The datasheet of the module is avaliable in the following link: https://www.seeedstudio.com/wiki/images/c/cd/Bluetooth4_en.pdf

randdusing, 2016-04-30:
Do you mind closing this issue and opening an issue on the plugin's repository. This most likely isn't an issue with the Angular wrapper.

MuriloACQ, 2016-04-30:
Sure, sorry for open it in wrong place

closed 2016-04-30 by MuriloACQ