randdusing/ng-cordova-bluetoothle #15: Problem using ng-cordova-bluetoothle

victorazevedo, 2016-05-05:
I´m having this error in Log when I try to connect:

Initialize Error:
{"error": "unsupported", "message" : "Operation unsupported"}

I´m using Motorola Moto X with Android version 6.0

I added <plugin name="BluetoothLePlugin" value="com.randdusing.bluetoothle.BluetoothLePlugin" /> in my config.xml and in my AndroidManifest.xml I have Bluetooth Permissions:

randdusing, 2016-05-07:
Are you calling bluetoothle.initialize before the deviceready event has been fired?

pngsantos, 2016-05-12:
Same issue. Using with ionic and already have it inside the $ionicPlatform.ready() wrapper.

randdusing, 2016-05-12:
What type of device?

victorazevedo, 2016-05-12:
Randdusing I used your example without change anything.
These examples worked with me using the cordova-plugin-ble-central: 

https://github.com/don/ionic-ble
(cordova plugin add https://github.com/don/cordova-plugin-ble-central --save)

https://software.intel.com/en-us/creating-a-bluetooth-low-energy-app

But I have to turn the GPS on to the bluetooth plugin work. Do you know why or how can i change this ?

randdusing, 2016-05-13:
Location services must be enabled on 6.0. If you open up Android Device Monitor, do you see any issues?

randdusing, 2016-05-13:
Actually why are you adding that to config.xml? You shouldn't need to add anything

MathGrelck, 2016-06-07:
I'm having this issue on both LG Nexus 4 Android 5.1.1 and iPhone 5s iOS 9.3.2.
I followed the instructions for the provided example and the app runs on both devices, but I get that same error when initializing (both in peripheral mode and central). Check it out in the ionic view, id: d5fc1047

randdusing, 2016-06-07:
Is the BLE plugin supported by Ionic View? http://docs.ionic.io/docs/view-usage

MathGrelck, 2016-06-07:
Ups, sorry about that! I'm obviously still learning...

randdusing, 2016-06-07:
No worries!

Martijn-Jansen, 2016-08-05:
i'm experiencing the same issue "Operation unsupported" on every call in the sample app. I'm using iphone 6. even the initialize is returning this error.
I find it very hard to find a good plugin that works on ios and android for advertising.
Some input would be great.
[xsanyware.ionic.ble.zip](https://github.com/randdusing/ng-cordova-bluetoothle/files/403411/xsanyware.ionic.ble.zip)

randdusing, 2016-08-14:
@Martijn-Jansen Looks like you are using the example provided. Can you try removing and then adding back the plugin? You should only see that message if the plugin fails to load.

victorazevedo, 2016-08-14:
An alternative plugin that worked for me :
https://github.com/don/cordova-plugin-ble-central

closed 2016-09-13 by randdusing

blandman, 2017-06-13:
I'm having this issue on iOS only, Android loads fine and is able to scan, iOS receives ` {error: "unsupported", message: "Operation unsupported"}`
Will try a few more tricks

blandman, 2017-06-13:
I was able to get around this by removing the plugin, cleaning in xCode, and reinstalling before running `cordova build ios`