randdusing/ng-cordova-bluetoothle #16: Read Long Characteristic

herfedos, 2016-05-05:
Hi, 
as graduation project, i'm developing an hybrid application for the monitoring of vital parameter, which requires Bluetooth Low Energy. 
I'm using ng-cordova-bluetoothle plugin and all the methods that I have used so far are working. 
Now I want to read the frame from the BLE card to plot charts, but I have a problem, the frame length is 100bytes and I found that the maximum that can be read is 22bytes. have you any idea how I can use the read() to read this frame? is that even possible ?
thank you,

randdusing, 2016-05-08:
What platform are you on?

randdusing, 2016-05-08:
Also, this is most likely not an issue with ng-cordova-bluetoothle. I'd close this bug and open one on the plugin's repository.

herfedos, 2016-05-10:
ok, thank you.

randdusing, 2016-08-14:
Closing due to previous comments

closed 2016-08-14 by randdusing