randdusing/ng-cordova-bluetoothle #22: Trigger callback after scanTimeout

mpedersen15, 2016-08-12:
Hi there,

It seems that using the scanTimeout parameter with startScan does stop the scan after the given time (I checked this using isScanning with angular's build in $timeout), but it seems that no callback is triggered when stopScan is called automatically after the timeout. Is this the expected behavior? Is there a way to trigger a callback without using the $timeout service (as this kind of defeats the purpose of the scanTimeout parameter, right?)?

Thanks in advance for the help and for the awesome angular wrapper of a great plugin!

randdusing, 2016-08-14:
@mpedersen15 It should return a callback like below. I'm using the example app, but I uncommented the scanTimeout line. Maybe post your relevant code and any output? Possibly the promise is getting resolved/rejected before the scanTimeout code completes? And the stopScan will only fire when the scan is still scanning, so double check to make sure nothing else is causing the scan to stop. Maybe scanTimeout should resolve with something like 'Scan already stopped' if it already stopped?

> Start Scan : {"services":[],"allowDuplicates":false,"scanTimeout":5000,"scanMode":0,"matchMode":2,"matchNum":1}
> Start Scan Success : {"status":"scanStarted"}
> Start Scan Auto Stop : {"status":"scanStopped"}

mpedersen15, 2016-08-14:
@randdusing Thank you for the quick reply! I will take a look at the example app again and see what I find that's different. I'll post some code if I'm still stuck.

mpedersen15, 2016-08-15:
@randdusing It looks like the issue was that I had a null first parameter to startScan. Following your example app and replacing this parameter with a callback function does allow me to receive a callback on scanStopped. Is this the intended purpose of the first parameter to startScan?

randdusing, 2016-08-16:
Yep, the parameters are successCallback, errorCallback and params

mpedersen15, 2016-08-16:
I think I may have explained poorly in my last comment. If the example app you have the following startScan call:
```
 $cordovaBluetoothLE.startScan(params).then(function(obj) {
      Log.add("Start Scan Auto Stop : " + JSON.stringify(obj));
    }, function(obj) {
      Log.add("Start Scan Error : " + JSON.stringify(obj));
    }, function(obj) {
      Log.add("Start Scan Success : " + JSON.stringify(obj));

      addDevice(obj);
    });
```
Your other examples have a null instead of the first callback function (the one that logs "Start Scan Auto Stop").
```
$cordovaBluetoothLE.startScan({services:[]}).then(null,
    function(obj) {
      //Handle errors
      console.log(obj.message);
    },
    function(obj) {
      if (obj.status == "scanResult")
      {
        //Device found
      }
      else if (obj.status == "scanStarted")
      {
        //Scan started
      }
    }
  );
```
 What exactly is the first callback function doing?

randdusing, 2016-08-19:
Ahh I understand now. The first example you listed provides a resolve callback since the auto scan timeout will resolve the promise when the scan is stopped. In the second example, startScan is called without the scanTimeout property, so it's impossible for it to ever resolve. It will only notify with scanStarted or scanResult or just reject/error. Hopefully that makes more sense.

mpedersen15, 2016-08-19:
Ah I see! Thank you very much for taking the time to explain!

closed 2016-08-19 by mpedersen15