randdusing/ng-cordova-bluetoothle #23: android.bluetooth.BluetoothAdapter.getBluetoothLeAdvertiser()' on a null object reference

MathGrelck, 2016-08-16:
Running the example app on android using `cordova run android`, I get the following error when trying to advertise in peripheral mode:
```
Start Advertising Error : "Attempt to invoke virtual method
'android.bluetooth.le.BluetoothLeAdvertiser 
android.bluetooth.BluetoothAdapter.getBluetoothLeAdvertiser()' 
on a null object reference"
```
The devices I tried running on is both a Samsung Galaxy S6 mini and the LG Nexus 5X. Same error.
I have tried re-adding the plugin. `initializePeripheral` + `addService` works fine, but when triggering `startAdvertising`, I get the error. See the console.log file attached.
I have a question more: where is the characteristic `"1234"` (default for addService, i guess?) set within the code?
I have another question (note: my knowledge of BLE is still limited, so excuse me if the question seem dumb, obvious or a non-issue); what is the BLE profile that is used? Is it one supported by Apple, or do I have to get a MFi license when releasing the app to the App Store?
Thanks!
PS: I also attached the project.
[console.log.zip](https://github.com/randdusing/ng-cordova-bluetoothle/files/419878/console.log.zip)
https://www.dropbox.com/s/67lut57hpe4mfec/test.zip?dl=0

randdusing, 2016-08-19:
I actually bought a Nexus 5X a few weeks ago and will double check that I can advertise on it.

The service is added in ```$rootScope.addService ``` function. This is just a 'random' service UUID that I came up with.

BLE doesn't require any licensing like BT Classic!

randdusing, 2016-08-19:
Oh, you need to call bluetoothle.initialize() as well! If I don't call that, I receive the same error. If I call it, it works fine.

MathGrelck, 2016-08-23:
Perfect! Thank you so much randdusing.

closed 2016-08-23 by MathGrelck