randdusing/ng-cordova-bluetoothle #25: Confusion regarding target SDK 23 and android 4.3

MathGrelck, 2016-08-23:
I'm not sure how to interpret the minimum requirements:
```
Android 4.3 or higher, Android Cordova library 5.0.0 or higher, target Android API 23 or higher
```
How is it possible to build an app using API 23 for a phone running Android 4.3?
Isn't the API tied to the Android version, i.e. API 23 for 6.0? (or 22 for 5.1.1 in my case)
My reference is simply the Android SDK Manager which list the build tools categorised as `Android 6.0 (API 23)` etc.
This confuses me (mainly because I'm completely new to mobile development).

I've tried setting  `<preference name="android-targetSdkVersion" value="23"/>` and
  `<preference name="android-minSdkVersion" value="23"/>` in the config.xml file, which result in apk install errors when deploying to the device.

randdusing, 2016-08-25:
Target API 23 just means the Android code can use code up to API 23, but it's still backwards compatible...assuming the plugin developer doesn't try to run newer API code on devices running older API.

Also, please post in the plugin repo itself in the future!

MathGrelck, 2016-08-28:
Thanks a lot.
I will!

closed 2016-08-28 by MathGrelck