randdusing/ng-cordova-bluetoothle #31: unable to get devices.

mnadeemmdsol, 2016-10-05:
Hi i am using this plugin and its working fine for initialize , but i start scan , no devices .

  $rootScope.startScan = function() {
    var params = {
      services:[],
      allowDuplicates: false,
      //scanTimeout: 15000,
    };

    if (window.cordova) {
      params.scanMode = bluetoothle.SCAN_MODE_LOW_POWER;
      params.matchMode = bluetoothle.MATCH_MODE_STICKY;
      params.matchNum = bluetoothle.MATCH_NUM_ONE_ADVERTISEMENT;
      //params.callbackType = bluetoothle.CALLBACK_TYPE_FIRST_MATCH;
    }

    Log.add("Start Scan : " + JSON.stringify(params));

    $cordovaBluetoothLE.startScan(params).then(function(obj) {
      Log.add("Start Scan Auto Stop : " + JSON.stringify(obj));
    }, function(obj) {
      Log.add("Start Scan Error : " + JSON.stringify(obj));
    }, function(obj) {
      Log.add("Start Scan Success : " + JSON.stringify(obj));

      addDevice(obj);
    });
  };

randdusing, 2016-10-07:
Are you on Android? Make sure you have the appropriate permission: https://github.com/randdusing/cordova-plugin-bluetoothle#startscan

vcboy, 2016-11-22:
i can't find devices too

randdusing, 2016-11-24:
@vcboy Do you have BLE devices that are actively advertising? If on Android, have you requested the proper permissions and enabled location services?